umit-iace/tool-libs #63: ADC Implementation

TanDro, 2023-05-04:
@jhalmen Kannst du dir bitte mal die Implementierung vom ADC in
[e053cf8](https://github.com/umit-iace/tool-libs/commit/e053cf862ba3f50bc4b053a8f23b6e947bb7551a) für
[227932f](https://github.com/umit-iace/labor-hvac/commit/227932f1bb0bdfecc0ab64a8f9d00e94dfec4bdc)
anschauen. Gibt da noch so ein zwei Punkte, aber vom Prinzip, würde dass so sinn machen oder hast du einen anderen vorschlag?

jhalmen, 2023-05-05:
Ich denke sowohl ADC als auch PWM wrappst du grad in einen zusätzlichen layer der mMn unnötig ist: `PwmDriver` und `AdcDriver`. Alles was der `PwmDriver` derzeit tut kannst du in die implementierung der `Hvac::set*Pwm(double)` stecken. Evtl kann man sogar überlegen ob ein `getpwm()` und `getticks()` + dem rangecheck nicht sogar in die `stm/pwm.h` gehört.
wie das mit dem `AdcDriver` derzeit gedacht ist versteh ich noch nicht.

zwecks ADC implementierung wuerd ich da das DMA komplett rausziehen und `stm/dma.h` verwenden, und dem ADC nur den `&dma.handle` ueberreichen um den `__HAL_LINKDMA` call ausführen zu können. Auch sollten die ganzen `HAL_NVIC_*` aufrufe in die `board.initClocks` implementierung zu den ganzen anderen, damit das alles an einem Ort bleibt und man nicht anfangen muss zu suchen.
und wenn wir dabei bleiben, dass die software den adc triggert (zb. ueber
```cpp
k.every(10, [](uint32_t, uint32_t) {
    board.adc.measure();
});
```
dann brauchen wir wahrscheinlich gar keine callbacks im ADC, evtl noch nicht mal im DMA. müsste man probieren.

für den ADC hätte ich mir was in die folgende richtung vorgestellt:
```cpp
struct {
//...
ADC::HW analog { ADC1 };
ADC::Channel ambient {{
    .hw = &analog,
    .channel = 0,
    .rank = ...,
    .sampleTime = ...,
    .pin = DIO {GPIO_PIN_0, GPIOA, GPIO_MODE_ANALOG},
}}; // registriert den channel 0 beim ADC
ADC::Channel pipe {{
    .hw = &analog,
    .channel = 1,
    .rank = ...,
    .sampleTime = ...,
    .pin = DIO {GPIO_PIN_0, GPIOA, GPIO_MODE_ANALOG},
}}; // registriert den channlel 1 beim ADC
} board {};

Support::Support() {
    // ...
    board.analog.init(); // erst nachdem alle channels registriert sind kann der ADC initialisiert werden (weil davor noch unklar ist wie viele/welche channels ausgelesen werden muessen)
};
```

alternativ (und vielleich schoener) fuer die channels in obigem struct:
```cpp
    ADC::Channel ambient = analog.getChannel({
        .channel = 0,
        //....
    });
```
falls das funktioniert.

und letztendlich verwenden dann richtung 
```cpp
void Hvac::updateState(uint32_t, uint32_t) {
    state = {
        .pipe = board.pipe.get(),
        .ambient = board.ambient.get(),
        .heater = state.heater,
        .fan = state.fan,
    };
}
void Hvac::setFanPwm(double val) {
    state.fan = fmin(1, fmax(0, val));
    board.fan.pwm(state.fan);
}
void Hvac::setHeaterPwm(double val) {
    state.heater = fmin(1, fmax(0, val));
    board.heater.pwm(state.heater);
}
```

TanDro, 2023-05-07:
Passt, glaube bis auf die Zuweisung des Buffers zur Channelvariable sollte das jetzt passen.

jhalmen, 2023-05-08:
Hab's mir jetzt kurz angeschaut, https://github.com/jhalmen/tool-libs/commit/6cf96eb173b9cd52e343473f6666cd14a3f4d204 ist das Ergebnis, da ich keine rechte hab in diesem repo nun erstmal auf meinem eigenen branch. habe auch hvac upgedatet dazu: https://github.com/umit-iace/labor-hvac/commit/06171c63a19316f535e1351a460debff5936e1da

compiliert, bin mir aber ein bisschen unsicher in der ADC/DMA interaktion, nach meinem verständnis der datenblätter und der HAL war es bisher so dass tatsächlich kontinuierlich umgewandelt wurde, habe versucht das zu ändern, muss aber getestet werden. evtl muss
https://github.com/jhalmen/tool-libs/blob/6cf96eb173b9cd52e343473f6666cd14a3f4d204/stm/adc.h#L53
und
https://github.com/jhalmen/tool-libs/blob/6cf96eb173b9cd52e343473f6666cd14a3f4d204/stm/adc.h#L61 geändert werden

jhalmen, 2023-05-22:
ADC sollte mit [5ab2f0a](https://github.com/umit-iace/tool-libs/commit/5ab2f0a1ea65f7b14c642d12fe7cbd478a3e6312) fertig sein

closed 2023-05-22 by jhalmen